tjclement/esp-dimmer-hardware #3: What GPIO is used to control PWM?

ArnieX, 2016-12-13:
Hi @tjclement,

which GPIO is used for PWM?

I have just built it myself and it is awesome little board!

[image]

Thank you,

Martin

tjclement, 2016-12-14:
Hi @ArnieX,

Your module looks great, nice job! 

You can check out the example software I wrote for this board [here](https://github.com/tjclement/esp-dimmer-software/). The GPIO used is the ESP's pin 13; if you're using nodeMCU, [this pin is called pin 7](https://github.com/tjclement/esp-dimmer-software/blob/master/pwmServer.lua#L7).

Good luck, and have fun!

Best,
Tom

closed 2016-12-14 by tjclement

ArnieX, 2016-12-14:
Hi @tjclement,

thank you for the quick response. Unfortunately my module had died before I was able to do some serious things. I have no idea what happened and today I fly out to business trip. Have you had similar issue yourself? Now it doesn't even light the strip when connected nor start ESP up. Could that be fried mosfet? How do you exactly set the voltage? I did it to 3.30xV. Do you glue the screw so it couldn't accidantaly change?

Thank you for some hints.

Martin

tjclement, 2016-12-18:
The ESP datasheet mentions a maximum input voltage of 3.6v, so I set it to ~3.4v. Anecdotally, this increases the chip's stability. Your setting of 3.3v should be fine though, and is definitely not the cause of it not functioning anymore.

I've had failing ESPs on a few of my modules, mostly because of incorrectly set voltage converter settings. Like you said, putting a drop of hot glue on the screw of the voltage converter helps with this problem. I've also had a software issue once that made the device enter a boot loop, which I didn't notice for several hours. This increased its temperature so much it must have overheated. It was unresponsive afterwards.

If the blue LED on the ESP still blinks when you power on your module, the MOSFET or any of the PCB traces may have been fried. If the blue LED doesn't turn on at all, check if the voltage converter still outputs enough volts. If the blue LED remains on continuously, or the ESP gets very hot without doing anything, your ESP has been fried for some reason. If you find out what's wrong, please leave another comment. Perhaps we can improve the module design!

Also related, I'm working on a new version of this board that is even smaller and has 3 channels instead of 1. This allows for dimming traditional RGB LED strips. I've not pushed this to the master branch yet because the design is untested, but I'll push them to a seperate branch when I have a moment. You may find it interesting to look at, too.

ArnieX, 2016-12-18:
Hi Tom,

here is the firmware https://github.com/ArnieX/esp-dimmer-firmware

BR,

Martin

RiRomain, 2016-12-19:
@tjclement I've already made a fork of your project and made an RGB driver out of your board :-) :
https://github.com/RiRomain/esp-dimmer-hardware 
I've already assemble some unit and it works great for more than a year now! Thanks you for your design.