tjclement/esp-dimmer-hardware #4: Programming interface

ArnieX, 2017-01-06:
Hi @tjclement,

good job with the new version, I will likely order it just to have few boards at home. Good to see that you have made contact pads for DC-DC converter much bigger. What am I missing thought are pins for programming the esp. Sometimes there is need to connect it by cable, at least for first flashing. Is it missing by design or did you forget it? Is there place to add it? I would definitely love at least contacts like on the first module but this time with 2.54 pitch so I can snap or solder 2.54 header directly without modification.

Best regards,

Martin

tjclement, 2017-01-08:
Hi @ArnieX,

The flashing pins on the original design didn't work that well for me. It took a lot of tries to flash the module, each time failing when it was around 90% done flashing.

That, combined with the fact I'd like to replace the ESP chip on the module if it ever breaks down, instead of having to bin the entire module, I ended up soldering 2mm headers on my modules like this:

[image]

The ESP can be placed on the headers without soldering, and when you want to (re)flash them, just put them on a separate programmer module that has the same exposing headers:

[image]

This setup worked very well for me in the first design, so I chose not to have any programming pins exposed in the second design.

What do you think?

ArnieX, 2017-01-08:
Hi @tjclement,

well on previous module it worked perfectly, at least for me. I would also add 3,3V IN so you do not need to connect 12V IN but otherwise it was OK. Problem with your ideas, illustrated by the images you share is that it is good for prototyping, but not really for final module with everything soldered. On our device (but it is a lot bigger) we have made accessible header for programming like this image shows.

[image]

It is similar to your solution but allow for ESP to be soldered directly on it, gives it also power from regulator on NodeMCU. So that you do not need to connect it to mains while flashing the firmware. From our experience ESP don't break that easily, actually we haven't yet have any failing.

Here is the scheme for our programming interface:

[image]

and if other UART USB2SERIAL is used instead of NodeMCU we have jumper to enable flash mode:

[image]

Maybe you could find a way how to get it on your module too, even-thought it is lot more crowded. Otherwise I have already ordered more transistors and the new mini360 power so I can order new modules to test them. BTW my colleague had a great idea to make the programming interface to fit female USB-A but it would need holes to fix it, contact pads could be bent to be soldered on surface contacts. Then you would just hack USB cable to connect to UART USB2SERIAL programmer.

[USB-A AliExpress](https://www.aliexpress.com/item/10PCS-USB-type-A-USB-mother-Curved-needle/32265708803.html?spm=2114.01010208.3.10.2UqEr3&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10068_10000009_10084_10083_10080_10082_10081_10060_10062_10056_10055_10037_10054_10033_10059_10032_10099_10078_10079_10077_10000012_426_10103_10073_10102_10000015_10096_10052_10053_10107_10050_10106_10051,searchweb201603_10,afswitch_5,single_sort_0_default&btsid=8af59f64-a330-4f4f-be2b-329c17f3dfab)

That is just one of the ideas. Nothing will be perfect but I would rather have something than nothing ;)

Let me know if I can do more for you,

Martin

ArnieX, 2017-01-28:
Thanks for the idea with flashing socket from NodeMCU, it works flawlessly.

[image]
[image]

closed 2017-04-12 by tjclement