tjclement/esp-dimmer-hardware #5: Copper on PCB

Aertbei, 2017-04-12:
Is it necessary to order the PCB with 2 oz copper or is 1 oz fine?

tjclement, 2017-04-12:
1 oz is fine, with this you should have a theoretical limit of 8A of power before the traces reach temps of over 65 degrees Celsius.

If you go with 2 oz, you'd even be able to approach the 15A limit of the FDS8896 mosfet. This is cool, but by no means necessary.

Thanks for reaching out, have fun!

closed 2017-04-12 by tjclement

tjclement, 2017-04-12:
If you plan on having large currents run through the module, by the way, be sure to use the new triple channel design. This has beefed up traces that allow for the 8A I mentioned above.

The previous single channel design is rated for much less (~2.5A).