tjclement/esp-dimmer-hardware #9: Add microwave motion detector and brightness/color sensor

renne, 2018-04-12:
I suggest to add a 0,40 $ [RCWL-0516](https://github.com/jdesbonnet/RCWL-0516) microwave motion detector to a GPIO of the ESP8266 and an 2,- $ [APDS-9960](https://cdn.sparkfun.com/datasheets/Sensors/Proximity/apds9960.pdf) brightness/color sensor via I2C.

That way the LED-controller can switch the light on if there is movement in the room. The [APDS-9960](https://cdn.sparkfun.com/datasheets/Sensors/Proximity/apds9960.pdf) can be used to detect current brightness and color temperature in the room for closed-loop brightness/color control.

The gesture support of the [APDS-9960](https://cdn.sparkfun.com/datasheets/Sensors/Proximity/apds9960.pdf) can be used for dimming and changing the color temperature.

[APDS-9960 Arduino-library](https://github.com/sparkfun/SparkFun_APDS-9960_Sensor_Arduino_Library/tree/V_1.4.2)

tjclement, 2018-04-12:
Sweet idea. I'll expose the necessary GPIO pins on the next iteration of the module. Thanks ;)

hnykda, 2021-05-30:
:wave: wanted to ask if you plan to ever update it or not anymore? :upside_down_face:

tjclement, 2021-05-30:
I might, if enough people are interested in a new version. I've learned a lot in the 5 years since I made this design, so I'm sure it could be improved even further. :)

closed 2022-11-21 by tjclement