tjclement/esp-dimmer-hardware #11: MOSFET gate resistor position in Version 2 schematic

danricho, 2020-07-02:
I believe that your schematic needs correction in the version 2.
They show as in series between the ESP and the mosfets, but I think they are meant to be pull-down.

I have made my own version with a few additions [here](https://github.com/danricho/ESP8266_12VLED_Driver/).

Great design and thanks!

tjclement, 2020-07-02:
Hey, thanks for reaching out! The resistors there are current-limiting to prevent damage to the ESP should the mosfet fail closed. There should also be pulldowns to prevent flickering during booting of the ESP, you're right. During normal operation this doesn't matter too much, since the GPIOs are actively driven.

Your version looks good, nice work! I might update my design sometime in the future too.

closed 2020-07-02 by tjclement

danricho, 2020-07-02:
Fair enough. The first version of my PCB had the inline resistor only and the output wasn’t stable... I’m unsure what I did wrong as it was exactly the same as your PCB schematically... and I didn’t have that issue with yours (which I used for a year+). The updated version I’ve shared, seems to work but I didn’t consider the failure modes of the MOSFET!

Again, thanks for sharing your design!